RobotWebTools/ros3djs #90: Publishing image messages

daniel86, 2015-01-13:
Hi!
First of all, I am npot sure if this is really an issue or if I am doing something wrong.
I am trying to publish the pixel data of the WebGL canvas via a ROS message with
type 'sensor_msgs/Image' and I am trying to display these messages in RVIZ
but I was not able to figure out yet how that can be done.
My intention is to create a video that shows the GL canvas over time for particular
experiments with a PR2 robot.

I try to read the pixel values using glReadPixels:
```
var gl = rosViewer.renderer.getContext();
var width  = gl.drawingBufferWidth;
var height = gl.drawingBufferHeight;
var imageData = new Uint8Array(width * height * 4);
```
The content of ```imageData``` looks good.

Afterwards, I generate and send a message as follows:
```
var currentTime = new Date();
var secs = Math.floor(currentTime.getTime()/1000);
var nsecs = Math.round(1000000000*(currentTime.getTime()/1000-secs));
var msg = new ROSLIB.Message({
    header: {
        stamp: { secs:secs, nsecs:nsecs },
        frame_id: "image",
        seq: frame_counter
    },
    height: height,
    width: width,
    encoding: "rgba8",
    is_bigendian: 0,
    step: width*4,
    data: imageData
});
frame_counter += 1;
topic.publish(msg);
```

This yields in following error message:
```
[ERROR] [WallTime: ...] [Client 1] [id: publish:/canvas/snapshot:345] publish: field data must be a list or tuple type
```

So it seems that something went wrong during serialization of the ```data``` field.
Does someone has an idea what's going wrong?

rctoris, 2015-01-13:
My guess is that `Uint8Array` is an object, not an array like `data` expects. That is, it gets serialized into JSON incorrectly when publishing. Try taking the values from `Uint8Array` and placing them into a raw JavaScript array as your `data` value.

sevenbitbyte, 2015-01-13:
[Here is an example which uses an array instead](https://gist.github.com/sevenbitbyte/9cee45bdbfa5b064cf17)

daniel86, 2015-01-14:
Thanks for your quick response.
Unfortunately I run into the same problem (field data must be a list or tuple type) when using following code:
```
      var buf = new Uint8Array(width * height * 4);
      gl.readPixels(0, 0, width, height, gl.RGBA, gl.UNSIGNED_BYTE, buf);
      
      var pixels = [];
      for (var i = 0; i < buf.length; i++) {
          pixels.push(buf[i]);
      }
```

@sevenbitbyte: How do you obtain the context where getImageData is defined?
I am using ROS3D.Viewer which creates a THREE.WebGLRenderer instance where I get
the context from (THREE.WebGLRenderer.getContext()).

closed 2015-01-14 by daniel86