RobotWebTools/ros3djs #104: PR2 Meshes are not being loaded.

junkilee, 2015-03-03:
[image]

PR2 Meshes are no longer being loaded.
It still chooses ColladaLoader2, but the meshes are not being loaded for some reasons.
This use to work before.

rctoris, 2015-03-04:
What revision of ros3djs are you using? Does reverting to a different version change anything?

junkilee, 2015-03-04:
When I go back to 0.10.0, this problem doesn't occur. From 0.11.0 and up, this problem persists.

rctoris, 2015-03-04:
Strange looks like the changelog doesn't mention any collada changes in 0.11.0. Looks like just the materials aren't being applied correctly. No errors in the JS console?

junkilee, 2015-03-04:
I don't see any error logs at all. Console logs are clean. I've tried to change mesh image files to be jpeg, png.. None of them works.

rctoris, 2015-03-04:
Are you loading meshes from resources.robotwebtools.org?

rethink-rlinsalata, 2015-03-05:
I came across a similar issue with those versions of ros3djs and the Baxter URDF (https://github.com/RethinkRobotics/baxter_common/tree/master/baxter_description/urdf).

I recently updated an app from using version 0.9.0 to the latest version when I noticed it. I tried both 0.13.0 off the CDN and the sources from the 'develop' branch and saw the problem. I tried a bunch of different versions back to 0.9.0 to try to find where it broke and I spent some time trying to walk through the code and commits.

This is what it looked like with the latest versions from the 'develop' branch:
[image]

[image]

I forget how, but somehow I traced part of the problem down to the code related to materials, and after making these changes to the latest code on 'develop' (I believe I was essentially reverting a prior commit), I got it to look "better" though not completely right.

https://gist.github.com/rethink-rlinsalata/7c0b2350c43b4b27131f#file-material_hack-diff
```diff
--- /data/users/rlinsalata/dev/web_overlay/ver/develop/ros3d.js
+++ /data/users/rlinsalata/dev/web_overlay/ver/test_develop/ros3d.js
@@ -2742,26 +2742,18 @@
   for ( var l in links) {
     var link = links[l];
     if (link.visual && link.visual.geometry) {
-      // Save frameID
-      var frameID = tfPrefix + '/' + link.name;
-      // Save color material
-      var colorMaterial = null;
-      if (link.visual.material && link.visual.material.color) {
-        var color = link.visual.material && link.visual.material.color;
-        colorMaterial = ROS3D.makeColorMaterial(color.r, color.g, color.b, color.a);
-      }
       if (link.visual.geometry.type === ROSLIB.URDF_MESH) {
+        var frameID = tfPrefix + '/' + link.name;
         var uri = link.visual.geometry.filename;
         var fileType = uri.substr(-4).toLowerCase();
 
-        // ignore mesh files which are not in Collada or STL format
-        if (fileType === '.dae' || fileType === '.stl') {
+        // ignore mesh files which are not in Collada format
+        if (fileType === '.dae') {
           // create the model
           var mesh = new ROS3D.MeshResource({
             path : path,
             resource : uri.substring(10),
-            loader : loader,
-            material : colorMaterial
+            loader : loader
           });
           
           // check for a scale
@@ -2781,11 +2773,16 @@
             object : mesh
           });
           this.add(sceneNode);
-        } else {
-          console.warn('Could not load geometry mesh: '+uri);
         }
       } else {
-        if (!colorMaterial) {
+        var colorMaterial, shapeMesh;
+        // Save frameID
+        var newFrameID = '/' + link.name;
+        // Save color material
+        if (link.visual.material && link.visual.material.color) {
+          var color = link.visual.material && link.visual.material.color;
+          colorMaterial = ROS3D.makeColorMaterial(color.r, color.g, color.b, color.a);
+        } else {
           colorMaterial = ROS3D.makeColorMaterial(0, 0, 0, 1);
         }
         var shapeMesh;
@@ -2810,7 +2807,7 @@
         }
         // Create a scene node with the shape
         var scene = new ROS3D.SceneNode({
-            frameID: frameID,
+            frameID: newFrameID,
             pose: link.visual.origin,
             tfClient: tfClient,
             object: shapeMesh
```

It looked like this after, which is back to how it looked before (though if it the lighting / rendering is actually "correct" is questionable):
[image]

[image]

rethink-rlinsalata, 2015-03-05:
Here are shots of the exact same app/code/run but just iterating through all the versions using sources 0.9.0 - 0.13.0, develop, and "develop+my hack modifications":

(also, I was using these Viewer options:
```js
  var viewer = new ROS3D.Viewer({
    divID : 'urdf',
    width : 800,
    height : 600,
    antialias : true,
    intensity : 0.05,
    background : '#e2e7ea'
  });
```
)

0.9.0
[image]

0.10.0
[image]

0.11.0
[image]

0.12.0
[image]

0.13.0
[image]

develop
[image]

develop + material commit changes reverted hacky-mod
[image]

rethink-rlinsalata, 2015-03-05:
(p.s. in case you noticed the `intensity: 0.05` in the previous post), I've been having trouble trying to figure out the whole intensity and lighting settings, but I'm not sure if that's on my end or not.)

rctoris, 2015-03-05:
What versions of threejs and collada loader are you importing?

rctoris, 2015-03-05:
My guess is that the color material is overwriting the Collada's texture file. @rethink-rlinsalata does removing just https://github.com/RobotWebTools/ros3djs/blob/0.13.0/src/urdf/Urdf.js#L53 fix this for you? Not sure why the other edits would modify a Collada model given this `if` statement https://github.com/RobotWebTools/ros3djs/blob/0.13.0/src/urdf/Urdf.js#L42

rethink-rlinsalata, 2015-03-05:
The versions used in my first post (with the first picture and diff edit + resulting picture) are 'current' directly from cdn.robotwebtools.org/<library>/current/<library>.js -- so threejs:r61 and using the default loader, which I believe is ColladaLoader2:0.0.2.

In the second post, where I take a picture with each version of ros3djs, I got the versions from the README.md for each tag, which means:
0.9.0:
threejs: r56
ColladaLoader2: 0.0.1

and the rest are the same as current.

I'll try that line change.

rethink-rlinsalata, 2015-03-05:
Yes - it is fixed if I just remove that line (https://github.com/RobotWebTools/ros3djs/blob/0.13.0/src/urdf/Urdf.js#L53) (starting with the code as it is in 'develop').  It looks like this:
[image]

(which gives me a minimal diff with the original)

junkilee, 2015-03-06:
I also debugged this issue a little bit.
When you actually set 'https://github.com/RobotWebTools/ros3djs/blob/0.13.0/src/urdf/Urdf.js#L53', what happens inside the ROS3D.Urdf and ROS3D.MeshResource is that it first loads the .dae file using the ColladaLoader2 or ColladaLoader and then after loading it rewrites material variable for all the nodes in the scene (loaded by the ColladaLoader). (https://github.com/RobotWebTools/ros3djs/blob/0.13.0/src/models/MeshResource.js#L59).
The image texture file information is also saved in material variable and when it is overwritten, all the texture information loaded by the ColladLoader is gone.
This is because inside PR2 urdf file, both the mesh file information and the color information is set, therefore for every component, it contains both the mesh file link and its color information as material attribute.
pr2_description/urdf/base_v0/base.urdf.xacro
pr2_description/urdf/materials.urdf.xacro

My suggestion is that when the material variable in each node already exists. I think only the color information has to be rewritten instead of switching the material variable itself.

rctoris, 2015-03-09:
Will add a flag to use URDF textures that defaults to false. Will reference this issue in the PR patch.

rctoris, 2015-03-09:
https://github.com/RobotWebTools/ros3djs/commit/7fe924c78969f34ff773517bf4be48ad44ac6887 should fix it. This is released as part of 0.14.0 just now. You will also need the latest (0.13.0) version of roslib to test this. Let me know if this works.

rethink-rlinsalata, 2015-03-18:
I verified that 7f3924c did fix this issue for Baxter - it looks back to how it used to look now.  Sorry for the delay in getting back to you.

Also, Thanks for being so on top of fixing this and being a great maintainer!

rctoris, 2015-03-24:
Awesome, thanks!!

closed 2015-03-24 by rctoris