RobotWebTools/ros3djs #110: Tutorial code uses different ColladaLoader

PeterMitrano, 2015-03-22:
The working demo uses ColladaLoader instead of ColladaLoader2
When I used ColladaLoader2 it works, providing I have 

    loader :  ROS3D.COLLADA_LOADER

When I use 

    loader :  ROS3D.COLLADA_LOADER2

or when I use no loader at all carl doesn't show up.

rctoris, 2015-03-22:
That's because the tutorial was written for the PR2 which only works with colladaloader2. CARL (http://wiki.ros.org/carl_description) uses colladaloader. I think the solution is to find a working indigo example and update the tutorial. Perhaps it can even be CARL!

T045T, 2015-04-26:
Either way, I think the tutorial should document the fact that there are two Collada loaders to choose from. I've changed it to include both loaders and explicitly set ```loader : ROS3D.COLLADA_LOADER_2```.
Neither is strictly necessary, but I updated the tutorial text to point out that there is more than one loader, and different robot models may require different loaders.

As for CARL, I'm not entirely sure, but it seems like the model takes quite a bit longer to load than the PR2 model, and the resulting 3D visualization is slower as well. Otherwise, since the CARL files are also hosted on ```resources.robotwebtools.org``` it should be a drop-in replacement.

goretkin, 2017-08-25:
What is ColladaLoader2? From the URL of the CDN and from the contents, it appears to be

from this project: https://github.com/gero3/ColladaAnimationCompress/

but THREE.js (in the examples folders) also has a ColladaLoader2:

https://github.com/mrdoob/three.js/blob/5ce11d18c400791da0c9fbb4b6b9fcde2f754443/examples/js/loaders/ColladaLoader2.js

but that is probably too recent to be what's in the tutorials.