RobotWebTools/ros3djs #112: Mesh color is not displayed

sonyaa, 2015-05-05:
I'm using Interactive Markers with a mesh, and the mesh has a color. It used to work fine, but something has changed over the past couple months (I noticed there was a color-related commit recently, could that be it?), and now the meshes are all grey.

T045T, 2015-05-26:
Thanks for reporting this!

Two things would be really helpful to track this down:
First, can you pinpoint the version that broke your meshes? If you're using the CDN to include ros3djs, you can change to older versions by simply replacing ```current``` with a version number in the URL, like so: ```http://cdn.robotwebtools.org/ros3djs/0.14.0/ros3d.js```

Second, is it possible for you to share the problematic Mesh? That would make it a lot easier to reproduce the error. Barring that, just telling us the type of mesh (DAE / STL) already helps to narrow down the search for the bug.

sonyaa, 2015-06-01:
I'm using DAE meshes for the PR2 grippers (`package://pr2_description/meshes/gripper_v0/gripper_palm.dae` and so on).

If I switch to `http://cdn.robotwebtools.org/ros3djs/0.13.0/ros3d.js`, the color works - if I switch to `0.14.0` or `current`, all the meshes are grey again.

T045T, 2015-06-03:
Thanks, I'll look into it this weekend :)

DLu, 2015-08-13:
@T045T Any progress on this bug?

T045T, 2015-08-13:
I'm sorry, I thought I posted an update a while ago :/

I wasn't able to reproduce the problem with either version of ros3d.js, sadly.
Also, I won't have much spare time in the next few weeks. If someone else wants to tackle this, go ahead :)

DLu, 2015-08-23:
I have reproduced the error. [source code](http://metrorobots.com/mesh_color_test.tar.gz)

In Rviz:
[image]

In Browser:
[image]

Switching collada loaders does not seem to help.

DLu, 2015-08-23:
Okay, you can fix it by uncommenting this line: 
https://github.com/RobotWebTools/ros3djs/blob/develop/src/models/MeshResource.js#L66

It looks like @rctoris added this so that the code "only adds material to a model that has none". But sometimes you (and I) do want to overload the mesh's natural color. 

At least for the Markers, it only sets the material if the color is not all 0s. 
https://github.com/RobotWebTools/ros3djs/blob/develop/src/markers/Marker.js#L274

That seems reasonable enough to me. Are there other places we should include that logic, @rctoris ?

rctoris, 2015-08-24:
Maybe something similar should happen here in that case? https://github.com/RobotWebTools/ros3djs/blob/develop/src/urdf/Urdf.js#L40-L43

DLu, 2015-08-24:
The tricky part is the Marker msg always has a color field, so checking for all zeros serves as a proxy for seeing if the color is valid.

rctoris, 2015-08-24:
Hmmm, maybe revert then and if you don't want to overwrite the color in the URDF, then remove the tag?

DLu, 2016-03-08:
[Updated source code](http://metrorobots.com/mesh_color_test.tar.gz)

So there's still work to be done here. Here's the test I was running in RViz. 
[image]
Foreground: URDF with a regular mesh link, a mesh link with a blue material and a box object with a blue material. 
Background: Four markers:
 * one with the color set to all zeros and use_embedded_materials=True
 * one with the color set to blue with alpha=0.5, use_embedded=True
 * one with the color set to blue, alpha=1.0, use_embedded=True
 * one with the color set to all zeros and use_embedded_materials=False

Note that rviz is arguably wrong here in that the front middle link is not blue. 

Here's what it looks like in RWT (using current release)
[image]
Robot model matches rviz, markers are all using the embedded texture. 

Here's what it looks like with devel RWT
[image]
The robot model matches what you might expect. The second marker is 50% transparent, not 50% blue and 50% the embedded colors. The third marker is now correct. The fourth marker should not display the embedded mesh. 

It's clear at the very least the Marker object should check the use_embedded_mesh flag. The "correct" interpretation for the second marker is open to interpretation, and is probably fine.

rctoris, 2016-03-10:
Yeah I held back this release because some of our files at WPI were no longer colored correctly. This is an awesome test case @DLu !

jihoonl, 2018-01-04:
I think this has been addressed by #187.

closed 2018-01-04 by jihoonl