RobotWebTools/ros3djs #114: map and gmapping

rbonghi, 2015-06-11:
Hi!

In this days I'm using this library with gmapping, but I have a problem to show the realtime map:

[image]

In the function `OccupancyGrid` line [68](https://github.com/RobotWebTools/ros3djs/blob/develop/src/maps/OccupancyGrid.js#L68) you have set:
```javascript
  // move the map so the corner is at 0, 0
  this.position.x = (width * message.info.resolution) / 2;
  this.position.y = (height * message.info.resolution) / 2;
```
but I think is correct to set the `this.position.x` and `this.position.y` to zero.

If this issue is correct I can fix with new position reference.

T045T, 2015-06-11:
I believe ros3djs' behavior is correct.
Looking at your screen shot, the corner of the map is at (0,0), where it should be. the ```OccupancyGrid``` message does not support negative indices (its data is just an array), so it makes sense its origin would be at the corner rather than in the center.

rbonghi, 2015-06-12:
Hi T045T, 
we have a differente system references, because the (0,0) picture position is on the corner, but is different compare to (0,0) map reference, in this case is centred on the center of the picture.

In the first picture I've highlighted the axes for the map reference (picture taken in rviz) in the last picture we see the wrong (0,0) system picture reference.
[image]

I think to fix this problem to use the information message from topic **/map_metadata**, these informations are collected in `message.info`

```javascript
  this.position.x = (width * message.info.resolution) / 2 + message.info.origin.position.x;
  this.position.y = (height * message.info.resolution) / 2 + message.info.origin.position.y;
```
Finally, in my honest opinion, this is the solution for this problem
[image]

but I don't understand why the map and the grid does not have same dimension. I'll try to investigate about this.

rctoris, 2015-06-12:
You are correct that the position of the map should be pull from the meta data (it is actually a member of the OccupancyGrid itself so you wouldn't need to explicitly subscribe to map_metadata -- http://docs.ros.org/indigo/api/nav_msgs/html/msg/OccupancyGrid.html). Could you submit a PR with your bug fix?

As for the grid, it might have to do with the default dimension in ros3d differing from the rviz default? Just a guess though.

rbonghi, 2015-06-12:
Hi rctoris,
Yes, I know the information is in `message.info`

To solve all problem I would like to add the information about orientation collected in **map_metadata**

> As for the grid, it might have to do with the default dimension in ros3d differing from the rviz default? Just a guess though.

I'm not sure, because the size is not a multiple to the size of the grid. I think is a problem about approximation.

PR #116 (2015-06-12): Update OccupancyGrid.js
Fixed #114
commit 7053a9f3 (2015-06-12): Update OccupancyGrid.js

Fixed #114
closed 2015-06-12 by rctoris