RobotWebTools/ros3djs #127: How to control turtle_bot  or PR2 UDRF file in browser using keyboard tele_op

Shrawan99, 2015-09-04:
I want to access turtle_bot  or PR2 UDRF file running  in server and want to control from local network PC or web in that case I cant use joint_state_publisher GUI so want to use keyboard to tele-operate turtlebot UDRF. How can I do that.? I have made this system in Robot management system. I can access the GUI shown in attached image from local network pc as well as by port forwarding.

**How can I control UDRF running in browser using keyboard?**

[image]

rctoris, 2015-09-08:
The URDF is simply a visual display of the robot state on the ROS end. Without running some kind of simulation, physical robot, or custom node which changes the TF between two frames based on a ROS message (e.g., what http://wiki.ros.org/turtlesim is doing) you cannot control a URDF.

Shrawan99, 2015-09-10:
Dear sir, 
 Thank you for kind time to response my issue. But i wonder does the tutorials shown in this website *http://wiki.ros.org/keyboardteleopjs/Tutorials/CreatingABasicTeleopWidgetWithSpeedControl* works for my issues? or can we add the algorithm of joint_state_publisher to our html page like this *https://github.com/RobotWebTools/robotwebtools.github.io/commit/f0895d557541c03940aa7126884f7e471db594a3*
Thanking you for you time.

rctoris, 2015-09-11:
keyboardteleopjs simply sends velocity commands to a robot or rosnode which accepts them. The joint state publisher will also not solve your problem since there isn't a joint between the robot's base and the world. For you solution, you will need to make a custom ROS node which accepts velocity commands and publishes a TF from the robot's base to some fixed world frame. http://wiki.ros.org/tf/Tutorials

MerAARIZOU, 2016-03-07:
Hi, Sorry for interfering maybe it's not the main issue here. But what you're suggesting here @rctoris is making the urdf file "mimicking" the real robot. What if we do the opposite thing: manipulating the model and the real robot should mimicking it.

closed 2023-02-10 by MatthijsBurgh