RobotWebTools/ros3djs #143: How do I toggle objects visibility?

positron96, 2016-03-31:
It seems that I cannot toggle objects visibility programmatically anyhow. In this case, it would be a nice feature for interactive visualizations.

FuzzAU, 2016-04-01:
What about if you get access to the underlying three.js object and set it's visible flag?
What sort of objects were you trying to set visibility for? Maybe we can build an example.

positron96, 2016-04-01:
Hello. I've created two map objects like this:

        new ROS3D.OccupancyGridClient({
			ros : rosobj,
			topic : "/map",
			continuous : true, 
			offsetPose: new ROSLIB.Pose({position: {x:0,y:0,z:0.5} }), 
			rootObject : viewer.scene
		})

	new ROS3D.OccupancyGridClient({
			ros : rosobj,
			topic : "/move_base/global_costmap/costmap",
			continuous : true, 
			opacity: 0.5,
			offsetPose: new ROSLIB.Pose({position: {x:0,y:0,z:-1} }),     
			rootObject : viewer.scene
		})
	
I'd like to be able to toggle between them interactive to achieve rviz-like behavior.

Also, setting offsetPose for them does not seem to work. Opacity works strangely as well, but that's not that important.

FuzzAU, 2016-04-02:
I don't think there is an issue with ros3djs here.
It'd be a lot of work for this project to try and expose (and maintain) all of the underlying properties of the three.js objects in the scene that people might want to change. The ros3djs is a super readable codebase, and after a few minutes of poking around I've found out how to set the visibility of the occupancy grid client, and your issue with the offsetPose.

When the OccupancyGridClient is constructed it doesn't immediately build it's underlying three.js objects [OccupancyGridClient.js].(https://github.com/RobotWebTools/ros3djs/blob/develop/src/navigation/OccupancyGridClient.js)
Instead it does it when the map topic receives a message. As you can also see, it may also remove and rebuild the three.js objects when the map data changes.

Your options for setting visibility will have to make sure the threejs items inside the grid aren't null first, so:
`var map = new ROS3D.OccupancyGridClient({ ..
...
});
`
Then when you wish to change the visibility state:
`if(map.currentGrid) {
  map.currentGrid.visible = false; // set to what ever you'd like here
}
`

To account for the fact that you may not know if the currentGrid object gets rebuilt when a new map gets sent to the map topic:
`
map.on('change', function() {
   map.currentGrid.visible = false; // set to what ever you'd like here
});
`

Finally, if you check out the source you'll see the issue with offsetPose is that it is only taken in to account if you suffly a tfClient to the constructor of the occupancy grid client.

It might be useful to still have the offset apply even if the tfclient isn't being used. Anyone have an opinion on that?

FuzzAU, 2016-04-02:
@rctoris I suggest closing this as there is capability to do what is requested.

positron96, 2016-04-02:
Thank you! I will try the methods you suggested to make it work. But anyway, I think that object visibility is a very basic property and having a unified access to it for any class would be great. I certainly do not ask for all properties of underlying three.js objects.
As for offsetPose, it was not immediately clear for me that tfClient is required for it to work. Some documentation might be helpful here (for those do not know JS quite well, like me)

FuzzAU, 2016-04-02:
I agree that perhaps it might be best to at least update the wiki or code documentation for these fields as it is not immediately obvious

FuzzAU, 2016-04-02:
Out of curiosity @positron96 , did you check out the notes at the top of the OccupancyGridClient class?

positron96, 2016-04-02:
You mean the comment describing constructor arguments? Yep, I did.

rctoris, 2016-04-09:
Another issue is that not every ros3djs object is a single threejs object (e.g., arrows are two different shapes) so there would need to be some common API put in place to try and deal with toggling every object within ros3djs.