RobotWebTools/ros3djs #145: Tutorial on VisualizingAURDF not clear.

janga1997, 2016-05-28:
This [tutorial](http://wiki.ros.org/ros3djs/Tutorials/VisualizingAURDF) is not clear with respect to the path parameter to be specified in the creating the UrdfClient object. 
The path specified in this parameter replaces the **package://** in the URDF file when loading meshes. This needs to be specified, which leads to a lot of confusion for a newbie trying to use ros3Djs.

FuzzAU, 2016-05-30:
The path parameter doesn't replace package. Check out https://github.com/RobotWebTools/ros3djs/blob/develop/src/urdf/Urdf.js and you'll see what happens.
The package:// component is stripped off the the package and remaining file type information is appended to the path to get a URI to use for loading the resource.

I was thinking the other day that a simple flask webserver showing how to retrieve the URDF might actually be useful. I didn't find the path component too confusing. The documentation explicity says "Here we provide the Ros node object and TF client created above, the viewer's scene to render to, and a link to the base URL where out Collada files are too be loaded from. Note that due to browser security settings, these files must be hosted on a web server of some kind in order to be served properly.". As it says "base URL" it made sense to me. After building a little web server or stepping through the code you can quickly see what the requested URL is and adapt so it.

janga1997, 2016-06-01:
@FuzzAU Sure. When I figured it out, it made sense to me. But my point was that it wasn't mentioned clearly that the package:// component is stripped off, and replaced with the base URL. It might not make sense to a newbie, such as me. 
I made an edit to the tutorial recently regarding this.

closed 2017-01-23 by janga1997