RobotWebTools/ros3djs #146: How to visualize Nao robot?

zhaozj89, 2016-07-06:
If I have the mesh files of Nao in my computer, how to visualize it like PR2 in the tutorial? Thanks.

zhaozj89, 2016-07-06:
If I have the following urdf file:

`<?xml version="1.0" ?>

<robot name="nao" xmlns:xacro="http://ros.org/wiki/xacro">
  
  <material name="LightGrey">
    <color rgba="0.8 0.8 0.8 1.0"/>
  </material>
  <material name="Grey">
    <color rgba="0.6 0.6 0.6 1.0"/>
  </material>
  <material name="DarkGrey">
    <color rgba="0.3 0.3 0.3 1.0"/>
  </material>
  <material name="Black">
    <color rgba="0.1 0.1 0.1 1.0"/>
  </material>
  
  
  <link name="base_link"/>
  
  <link name="torso">
    
    <inertial>
      <origin rpy="0 0 0" xyz="-0.00413 0.00009 0.04342"/>
      <mass value="1.04956"/>
      <inertia ixx="0.00487953284" ixy="-0.00001428591" ixz="-0.00019545651" iyy="0.00470360698" iyz="0.00002224589" izz="0.0015671352"/>
    </inertial>
    <visual>
      
      <origin rpy="0 0 0" xyz="0 0 0.02075"/>
      <geometry>
        <cylinder length="0.2115" radius="0.015"/>
      </geometry>
      <material name="LightGrey"/>
    </visual>
  </link>
  <link name="HeadYaw_link">
    <inertial>
      <origin rpy="0 0 0" xyz="-0.00001 0.00014 -0.02742"/>
      <mass value="0.06442"/>
      <inertia ixx="0.00006223368" ixy="0.00000000042" ixz="0.00000007448" iyy="0.00006324431" iyz="0.00000010416" izz="0.00000549462"/>
    </inertial>
  </link>
  <link name="HeadPitch_link">
    <inertial>
      <origin rpy="0 0 0" xyz="-0.00112 -0.00003 0.05258"/>
      <mass value="0.60533"/>
      <inertia ixx="0.00208895741" ixy="0.00000549094" ixz="0.0001133995" iyy="1932222.63e-9" iyz="-0.00002803917" izz="0.00082257793"/>
    </inertial>
    <visual>
      <origin rpy="1.57 0 0" xyz="0 0 0.058"/>
      <geometry>
        <cylinder length="0.14" radius="0.04"/>
      </geometry>
      <material name="LightGrey"/>
    </visual>
  </link>
  <link name="gaze"/>
  <link name="CameraTop_frame">
  </link>
  <link name="CameraBottom_frame">
  </link>
  
  <link name="l_wrist"/>
  <link name="r_wrist"/>
  <link name="l_gripper">
  </link>
  <link name="r_gripper">
[... 488 lines cut ...]
    <axis xyz="0 1 0"/>
    
    
    
    <limit effort="30" lower="-1.772308" upper="0.484090" velocity="3.0"/>
  </joint>
  <joint name="RKneePitch" type="revolute">
    <parent link="RHipPitch_link"/>
    <child link="RKneePitch_link"/>
    <origin rpy="0 0 0" xyz="0 0 -0.1"/>
    <axis xyz="0 1 0"/>
    
    
    <limit effort="30" lower="-0.092346" upper="2.112528" velocity="3.0"/>
  </joint>
  <joint name="RAnklePitch" type="revolute">
    <parent link="RKneePitch_link"/>
    <child link="RAnklePitch_link"/>
    <origin rpy="0 0 0" xyz="0 0 -0.10290"/>
    <axis xyz="0 1 0"/>
    
    
    <limit effort="30" lower="-1.186448" upper="0.922747" velocity="3.0"/>
  </joint>
</robot>`

MerAARIZOU, 2016-07-06:
Put your mesh files in a http server, such as apache, 
in your URDF file load them from their URL (mesh =filename="package://mesh_URL"/>
`<mesh filename="package://nao_description/meshes/mesh1.dae"/>` (just an example)
then load your robot description into the memory from a launch a file:
`<launch>
	<param name="robot_description" command="$(find xacro)/xacro.py '$(find your_nao_package)/url_to_your_xacro_file/cyton_gamma_300.xacro'" />
	<node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" /> 
	<node name ="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
		<param name="use_gui" value="true"/>
	</node>
</launch>`
in the html file be sure to setup the path parameter:
` // Setup the URDF client.
      var urdfClient = new ROS3D.UrdfClient({
      ros : ros,
      tfClient : tfClient,
      path : 'http://URL_to_directory_containing_your_meshes_in_apache',
      frameID  : 'base_link',
      rootObject : viewer.scene,
      loader : ROS3D.COLLADA_LOADER_2
    });`
and after launching rosbridge_websocket.launch and running tf2_web_republisher you should visualize your model in a web page.
let me know if it works or if it there is any problem

zhaozj89, 2016-07-07:
:+1: Thanks. I am using nodejs` http-server`, and it works well too. Now I can see the nao visualization. Thanks.

tomeroto, 2018-04-30:
hi ghost, can you send me the mesh of Nao model and working project, I want to use that into my project too.

closed 2021-06-13 by MatthijsBurgh

PVGanesh2000, 2023-12-15:
How to visualize my turtlebot3 waffle on web, Thanks for the help