RobotWebTools/ros3djs #149: How to add three.js object in View?

zhaozj89, 2016-07-19:
Hi, I want to ask how to add three.js object in ros3djs View, or other webgl based js library,such as Babylon.js? Actually I want to plot a 3D heatmap for a robot model in View. Thank you!

FuzzAU, 2016-07-20:
Once you've created a ros3js viewer you can access the three scene by accessing the scene variable.

`
e.g.
// Create the main viewer.
var viewer = new ROS3D.Viewer({
     divID : 'markers',
     width : 800,
     height : 600,
     antialias : true
});

// Create a simple Three JS cube
var geometry = new THREE.CubeGeometry( 200, 200, 200 );
var material = new THREE.MeshBasicMaterial( { color: 0x000000 } );
var mesh = new THREE.Mesh( geometry, material );

// Add the threejs object to the scene inside ros3djs
viewer.scene.add(mesh);
`

zhaozj89, 2016-07-21:
:+1: Thanks!

T045T, 2017-02-17:
Looks like this has been cleared up. If you have further questions, don't hesitate to ask (preferably on answers.ros.org)!

closed 2017-02-17 by T045T