RobotWebTools/ros3djs #150: Interactive Markers ignoring Marker frame_id

velveteenrobot, 2016-07-27:
Interactive markers in ros3djs seem to have different behaviour than in Rviz.

I publish the following interactive marker:

```
markers: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: base_link
    pose: 
      position: 
        x: 1.0
        y: 1.0
        z: 1.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
    name: primitive_0
    description: ''
    scale: 0.20000000298
    menu_entries: []
    controls: 
      - 
        name: ''
        orientation: 
          x: 0.0
          y: 0.0
          z: 0.0
          w: 0.0
        orientation_mode: 0
        interaction_mode: 2
        always_visible: True
        markers: 
          - 
            header: 
              seq: 0
              stamp: 
                secs: 0
                nsecs:         0
              frame_id: ''
            ns: ''
            id: 2000
            type: 2
            action: 0
            pose: 
              position: 
                x: 0.0
                y: 0.0
                z: 0.0
              orientation: 
                x: 0.0
                y: 0.0
                z: 0.0
                w: 1.0
            scale: 
              x: 0.05
              y: 0.05
              z: 0.05
            color: 
              r: 1.0
              g: 0.0
              b: 0.0
              a: 1.0
            lifetime: 
              secs: 2
              nsecs:         0
            frame_locked: False
            points: []
            colors: []
            text: ''
            mesh_resource: ''
            mesh_use_embedded_materials: False
          - 
            header: 
              seq: 0
              stamp: 
                secs: 0
                nsecs:         0
              frame_id: base_link
            ns: ''
            id: 3000
            type: 2
            action: 0
            pose: 
              position: 
                x: 0.0
                y: 0.0
                z: 0.0
              orientation: 
                x: 0.0
                y: 0.0
                z: 0.0
                w: 1.0
            scale: 
              x: 0.1
              y: 0.1
              z: 0.1
            color: 
              r: 0.0
              g: 1.0
              b: 0.0
              a: 0.5
            lifetime: 
              secs: 2
              nsecs:         0
            frame_locked: False
            points: []
            colors: []
            text: ''
            mesh_resource: ''
            mesh_use_embedded_materials: False
        independent_marker_orientation: False
        description: ''
poses: []
erases: []
```

The interactive marker is published at (1.0, 1.0, 1.0) in base_link. One of the sub-markers is published at (0, 0, 0) with a frame of 'base_link' and one at (0, 0, 0) with an empty frame. In Rviz they display as below, which is the expected behaviour. One shows up at the base of the robot and one shows up at (0, 0, 0) relative to the parent interactive marker.
[image]

For ros3djs, both show up in the same place at (0, 0, 0) relative to the parent marker.
[image]

velveteenrobot, 2016-07-27:
The behaviour for my example is correct prior to adc02f2c0ec766ae0bb2ba13556074d45de20691. Although that commit probably fixes other Marker frame issues.

T045T, 2017-01-30:
fixed by merging #152 - for some reason Github is not showing that PR as merged, even though it is.

closed 2017-01-30 by T045T