RobotWebTools/ros3djs #154: No data received from rostopic /depthcloud_encoded

jacksonkr, 2016-10-26:
I can see rostopics on `localhost:8080` and I can even view a number of camera topics in realtime on the web:
```
http://localhost:8080/stream_viewer?topic=/camera/depth/image
http://localhost:8080/stream_viewer?topic=/camera/rgb/image_rect_color
http://localhost:8080/stream_viewer?topic=/camera/depth_registered/sw_registered/image_rect
```

The issue is `http://localhost:8080/stream_viewer?topic=/depthcloud_encoded`. If I view that stream then I see this output in the window where I ran `$ rosrun depthcloud_encoder depthcloud_encoder_node _depth:=/camera/depth_registered/image_rect _rgb:=/camera/rgb/image_rect_color`

[ WARN] [1477518125.161876892]: Messages of type 1 arrived closer (0.033357374) than the lower bound you provided (1.500000000) (will print only once)

arnasIV, 2017-02-07:
Have you found the solution of this problem?

T045T, 2017-02-07:
The warning isn't critical (though I'm not sure why the value 1.5 was chosen there) and shouldn't affect whether or not you see an image.

A few things you can try:
* Check the browser console for any error messages
* Check that the topic has information (i.e. ```rostopic echo -n 10 /depthcloud_encoded```)

Also, if you can point to a bagfile that reliably shows off the missing video stream, that would be great, since I don't currently have a Kinect to test this with.

arnasIV, 2017-02-07:
There are no errors in the browser console. The topic does not publish anything. Actually there is a problem with the callback, synchronizing disparity image and rgb image topics.

T045T, 2017-02-07:
If the topic doesn't publish anything, this is actually an issue with [depthcloud_encoder](https://github.com/RobotWebTools/depthcloud_encoder/).
Additionally, if the warning really indicates a problem with the callback, setting the Kinect frame rate to 0.5 (i.e. one frame every two seconds) should fix it. Of course, that's not a final solution, but it would confirm that violating that lower bound is our problem, rather than some other incompatibility.

Another likely point of failure is the bit depth of the depth topic. Can you confirm that the topic contains a single 32-bit channel?

JohanJarvi, 2021-04-21:
I know it's a bit late but I just ran into what seems to be a very similar issue and raised it in depthcloud_encoder github:
https://github.com/RobotWebTools/depthcloud_encoder/issues/16