RobotWebTools/ros3djs #155: showing URDF model in browser: all links are in origin

ppm-tahi, 2016-11-30:
Hi,

I am trying to use my robot model (nachi mz07) in ros3djs. I have the model in urdf format, rviz show it well, but when I am trying to show it in ros3djs in my browser, all of the links are located in origo. Plase see the attached pictures.

[image]

On the left: NACHI mz07 in the browser
On the right: the model in rviz

as you can see all of the things are loaded well (meshes), but the position is not so okay.
What should I do to fix this issue?

Here is my launch, how I am starting things (rosbridge is running from somewhere else!):
[image]

Thanks in advance!
Balint

T045T, 2016-11-30:
Hey Balint,

The HTML/JS code you use to show the URDF in the browser would be super useful to help debug (and reproduce your problem).

ppm-tahi, 2016-12-09:
Hey, 

here you can download it: 
https://www.dropbox.com/s/4aztkah7ei5bzui/nachi_mz07_01_support.zip?dl=1

Sorry but I had to ask that I can send it or not (it is not mine).

MerAARIZOU, 2016-12-09:
After launching all the packages, type in a terminal rosrun tf2_tools view_frames.py This is will create a pdf file showing the bases tree of your model. To give you an idea here's what mine looks like  
[image]
You notice that the root of that tree, at the very top is "odom" base. So in my HTML code, I'll put it as the fixedFrame parameter
`
// Setup a client to listen to TFs.
var tfClient = new ROSLIB.TFClient({
ros : ros,
angularThres : 0.01,
transThres : 0.01,
rate : 10.0,
fixedFrame : 'odom'
});
` 
Hope this will help

ppm-tahi, 2016-12-13:
Thanks, now it is working!

closed 2016-12-13 by ppm-tahi

msjulio00, 2018-07-21:
Hello @T045T.

Currently I'm trying to recreate this project but the html page doesn't show anything. I'm running the launch file showed above and rosbridge, apparently there are no errors in the web console and chrome can access the server where the meshes are but still the urdf is not there, I'm using the official urdf.html of ros3djs. I don't know what else could be wrong if you could help that would be great.

Thanks

sneakyc4k3, 2020-06-18:
I am having the same issue, and getting the error message `ERROR: cannot launch node of type [tf2_web_republisher/tf2_web_republisher]: Cannot locate node of type [tf2_web_republisher] in package [tf2_web_republisher]. Make sure file exists in package path and permission is set to executable (chmod +x)` even though I have the tf2_web_republisher file cloned from github inside already.