RobotWebTools/ros3djs #158: Update code base to work with modern three.js releases

sevenbitbyte, 2017-01-19:
three.js is currently at version 0.84.0 but ros3d requires version 61. Attempting to use a modern threejs version results in the below warnings and errors. There are likely many more minor tweaks require beyond these:

```
THREE.WebGLRenderer: .shadowMapEnabled is now .shadowMap.enabled.
THREE.Projector has been moved to /examples/js/renderers/Projector.js.
THREE.ImageUtils.loadTexture has been deprecated. Use THREE.TextureLoader() instead.
THREE.ShaderMaterial: attributes should now be defined in THREE.BufferGeometry instead.
THREE.ShaderMaterial: 'attributes' is not a property of this material.
THREE.ParticleSystem has been renamed to THREE.Points.
```

viktorku, 2017-01-26:
Take a look at https://github.com/Intermodalics/ros3djs/pull/1, which already bumps it up to r70 (among other things)...

sevenbitbyte, 2017-01-26:
@viktorku Awesome work, can't wait to leverage it. Any particular reason for targetting r70?

I've been dying to be able to fully tear down the WebGL context which is why I'm interested in moving up to r8x for better teardown support but maybe that made it into r70.

jihoonl, 2017-12-08:
I'm trying to update the codebase to work with version 0.88.0. The basic functions are getting ready now except interactive marker and urdf. Did you have any issue with highlighting the mouseover objects when you use r70?

jihoonl, 2017-12-08:
See #202

jihoonl, 2018-01-04:
Resolved by #202

closed 2018-01-04 by jihoonl