RobotWebTools/ros3djs #177: Removal of instances (viewer, URDFClient, TFClient)?

ninaschaaf, 2017-04-24:
We use the ros3D library for visualizing different robot models for different ROS-Connections in our web-app. 
Unfortunately this use-case produces a memory leak due to the necessity to create multiple instances of "Viewer", "URDFClient" and "TFClient".
Is there any way to "update" the existing instances, for instance when the ros-connection changes -  in order to avoid having to create new instances for each ROS connection? Or is there a way to remove/dispose the instances completely?

renarded, 2018-05-24:
Up ! :)

I have exactly the same issue as @ninaschaaf . I want to be able to remove the robot on the viewer when I disconnect from ros, and load the robot again (not necessarily the same model, depends on the robot I'm connected to) on another connection, with a new TFclient and URDFClient.

I have tried to set the Viewer, Urdfclient and Tfclient objects to undefined, and also to remove the <canvas> HTML tag inside the urdf div using removeChild(). With no success.

mjsobrep, 2019-08-29:
The way that I am currently handling this is to leave the viewer there. It isn't tied to ROS anyway. Then I remove the urdf from the Three.js scene before releasing the references to the tf client and urdf client. This seems to be working, but there could be a bug or leak that I am not seeing. 

This ends up looking this: 
```jsx
viewer.scene.remove(client.urdf);
setClient(null);
```
where the `viewer` is the `ROS3D.Viewer` instance, the `client` is the `ROS3D.UrdfClient` client and `setClient` just sets the variable that stores the client. I don't keep around the `ROSLIB.TFClient` since it is referenced in the urdf client.