RobotWebTools/ros3djs #180: Release new version.

jihoonl, 2017-06-28:
The current version is two years old...,

viktorku, 2017-09-18:
I would also release it on Yarn. And if possible, I would get rid of hosting the built (concatenated) and minified versions in version control. It only unnecessarily adds complexity and burden to contributors/maintainers when opening PRs. It's something that actually is frowned upon in the JS world (and beyond) :) The direct links on the CDN can stay tho.

my 2c

jihoonl, 2017-09-20:
Yeah I agree. I am thinking to check if travis can push the built module when a PR gets merged. https://github.com/RobotWebTools/roslibjs/issues/260

Or configure CDN to get updated when the repo is updated.

closed 2021-06-13 by MatthijsBurgh