RobotWebTools/ros3djs #185: What changes must be made at the meshes

JoonasWa, 2017-07-20:
hi everybody
I like to visualize the UR10 with ros3djs. So I have to put the meshes in a http server. I did this but the code can't visual the robot. I tried to put the meshes of the pr2 on a http server too and I had the same problem. So what do I have to change in the meshes? Where can I see the meshes from http://resources.robotwebtools.org/?
Thanks

jihoonl, 2017-07-25:
Instead of using the resources in our demo server, you can use your own web server to host meshes. Try to run a web server that uses pr2_description package as a root path.

MerAARIZOU, 2017-07-25:
Install a http server on you computer, (apache or whatever) put your HTML file on your server; let's say in /var/www/html/test/3d.html. put pr2_description or ur10_descprition folder in the same location as your html file (/var/www/html/test) and set the parameter path to 'http://localhost/test' of urdfClient. Restart your web server and it should work

closed 2018-01-04 by jihoonl