RobotWebTools/ros3djs #191: Marker text color not displayed

3Dmizu, 2017-08-01:
Hi.
I wrote this code that it shows 3D text on 3D viewer.
https://gist.github.com/3Dmizu/630a94d3f6bd6ae9cfdbbb8fea7f9504

But text color doesn't change on browser using ROS3DJS even though it show correctly on rviz.
[image]
[image]

Thanks.

jihoonl, 2017-08-01:
#188 should have fixed this issue. Can you try with the latest version?

3Dmizu, 2017-08-02:
Sorry I missed that post.
I have used REVISION:"0.16.0", so I replaced REVISION:"0.17.0-SNAPSHOT" and it works well.

Thank you.

closed 2017-08-02 by 3Dmizu