RobotWebTools/ros3djs #192: Marker text inside transparent cube not displayed

3Dmizu, 2017-08-02:
Hi.
Please excuse me for continuing post.
I wrote this code that shows 3D text in transparent cube.
https://gist.github.com/3Dmizu/d28955dbcfaac43cf2940188bd311d9d

Text is hidden by cube in ROS3DJS, but rviz works well.
It occurred in "REVISION : 0.17.0-SNAPSHOT".

[image]
[image]

Thanks.

jihoonl, 2017-08-30:
interesting.. maybe a bug in three.js?

viktorku, 2017-09-18:
It might be related to #172. Did you try debugging and inspecting the generated values for the objects' transparencies? I'll try to reproduce this issue with your sample gist.

T045T, 2017-12-17:
This is related to the way ThreeJS renders sprites. One easy solution is to not have depth tests for text markers, but that would mean they don't get occluded by anything ever.

A more nuanced solution would be quite a bit more work (and possibly a performance cost), but I'll keep looking into it.