RobotWebTools/ros3djs #197: How do I use this library properly using es6 import - npm

reinzor, 2017-10-30:
I tried installing this package using npm:

    npm install --save git://github.com/RobotWebTools/ros3djs.git#develop

and then import

    import ROS3D from 'ros3d'

However, this is not working, can somebody help?

rivertam, 2017-11-20:
You should really give more details in your issues. Are you getting an error message? Is it not compiling? Are you using a compiler like webpack? What node version are you using? Generally, if an issue contains "not working" in it, it's not giving enough detail.

That said, I can spot right away you're importing 'ros3d' instead of 'ros3djs' which is at least one problem.

sevenbitbyte, 2017-11-23:
Hi @reinzor just wanted to point out that the package is released to npm at: https://www.npmjs.com/package/@robot-web-tools/ros3djs

Please let me know if you encounter any issues with the npm release we've just started publishing there.

jihoonl, 2018-01-18:
Can try `npm install ros3d?`. I have confirmed that it downloads modules.

[image]

bvobart, 2018-01-19:
@jihoonl Yes, it downloads modules, but when importing the module, the imported object stays empty. I have gone through the trouble of creating *two* minimal working examples: one with create-react-app as I mentioned in my previous comment, and the other as basic as possible plain JavaScript launched with NodeJS. Both show the same problem.

Try it yourself: https://github.com/bvobart/ROS3D-NPM-MWE

keego, 2018-02-28:
@bvobart I also have the same problem. It looks like the main issue is [how the ROS3D module is exported](https://github.com/RobotWebTools/ros3djs/blob/develop/src/Ros3D.js#L6), namely that it *isnt*. Simply `export`ing `ROS3D` worked for me but that's a breaking change. 

@sevenbitbyte @jihoonl Would you guys be down to follow the recommended `pkg.module` route [suggested by the guys over at Rollup](https://github.com/rollup/rollup/wiki/pkg.module)? If so I already have the recommended changes tested and in my forked branch, which I can gladly throw into a PR :)

sevenbitbyte, 2018-02-28:
@keego Whoops great catch on the missing export! if you already have a fix please PR that so we can get this fixed for everyone.

The rollup approach looks like the right future proof thing to do to me. Seems like both changes should not have any adverse affects as this is the first time ros3djs has been pushed to NPM and clearly has not been usable as of yet.

closed 2018-07-03 by jihoonl