RobotWebTools/ros3djs #199: CDN links not working

sausax, 2017-11-05:
I am getting access denied when trying to use the CDN links

dkorobov2, 2017-11-09:
I have the same problem...

jihoonl, 2017-11-13:
RWT CDN and demo server are currently experiencing hosting issue. I suggest to host them locally or directly from github for now. I will post in the mailing list as soon as I get some conclusion.
See this thread : https://groups.google.com/d/msg/robot-web-tools/PlW1ECoL91Y/1ss6BPp_DAAJ

jihoonl, 2017-11-20:
ah yes.. I couldn't find the same version that we uploaded in CDN..

Faibk, 2017-11-20:
It seems like ColladaLoader2 has become the only ColladaLoader on three.js, when I just use the loader bundled with three.js everything seems to work fine...

Faibk, 2017-11-20:
That's the one btw:
[colladaloader2.zip](https://github.com/RobotWebTools/ros3djs/files/1488141/colladaloader2.zip)

jihoonl, 2017-11-24:
Awesome thanks Faibk.

jihoonl, 2018-01-04:
All links should work as of now. I have uploaded all updated modules in the new cdn(https://static.robotwebtools.org/xxxx).

Reopen this issue if anything is missing.

closed 2018-01-04 by jihoonl

duwudi, 2021-01-10:
Seems the CDN links are dead again, will this be resolved or should I just load them locally?