RobotWebTools/ros3djs #204: Meshes turned with develop branch

asoriano1, 2017-12-21:
Hi, 
I was using the 0.17.0 revision and my URDF model was loading correctly.
Now I have moved to develop branch and each component of the URDF model is well positioned but they are turned 90 degrees.
Any idea?
Thanks in advance and best regards

jihoonl, 2017-12-21:
Hi @asisbot, yeah it was actually because ROS robots don't follow the proper collada standard format. Here we added custom collada loader that can load model just like rviz. Try to import this [ColladaLoader](https://github.com/RobotWebTools/ros3djs/blob/develop/examples/js/ColladaLoader.js) after ros3d.js import line.

asoriano1, 2017-12-21:
Yes, importing the ColladaLoader after ros3d.js the URDF is shown correctly.
Thanks @jihoonl. Best regards.

closed 2017-12-21 by asoriano1