RobotWebTools/ros3djs #211: none of the examples work

jtoy, 2018-01-23:
http://robotwebtools.org/demos.html#ros3djs

are they supposed to work?
is this project dead? 
chrome version on mac high sierra: Version 63.0.3239.132 (Official Build) (64-bit)

T045T, 2018-01-23:
Good news first: The project is very much noch dead!

Bad news: The servers behind those demos were migrated a while ago, and apparently not every necessary part made the move. Also, some demos still use the old `cdn.robotwebtools.org` url instead of `static.robotwebtools.org`

@rctoris, can you take a look at this?

jihoonl, 2018-01-23:
the project is still live. Just demo server is not up and running yet. If you run examples locally, all should work.

jtoy, 2018-01-24:
great to hear,  I think its good to get back up so people can see the potential!

brean, 2018-04-13:
Are there any plans to provide a websocket-server again?

keego, 2018-05-24:
Any updates on the when the examples will be working again? All the demos still either complain about there being no websocket
```
WebSocket connection to 'ws://demo.robotwebtools.org:9090/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
```
or still point to the incorrect cdn
```
GET http://cdn.robotwebtools.org/...
```

pleconte87, 2019-08-23:
what are the plans?

MatthijsBurgh, 2023-02-07:
Examples are updated

closed 2023-02-07 by MatthijsBurgh