RobotWebTools/ros3djs #224: How to load Collada files from local files ?

renarded, 2018-05-14:
First of all, I'm aware of what's written in the [URDF tutorial](http://wiki.ros.org/ros3djs/Tutorials/VisualizingAURDF) 

> Note that due to browser security settings, these files must be hosted on a web server of some kind in order to be served properly 

Still, I want to be able to load Collada files from local files. I am developing an electron application, my goal is to package the app with the Collada files inside, and load them directly instead of using an HTTP server.

I've successfully run the "Visualizing a URDF" tutorial using my own Collada files served by a local HTTP server.

Here's the code for the UrdfClient : 

```javascript
    // Setup the URDF client.
    var urdfClient = new ROS3D.UrdfClient({
      ros : ros,
      tfClient : tfClient,
      path: 'http://localhost:8000/',
      rootObject : viewer.scene,
      loader : ROS3D.COLLADA_LOADER
    });
```

With the HTTP server launched from /home/myprofile/catkin_src/src/ on port 8000.

I've tried using 
- path: '/home/myprofile/catkin_src/src/',
- path: 'file://localhost/home/myprofile/catkin_src/src',
- path: 'file:///home/myprofile/catkin_src/src',

None of them work, the scene on the screen is empty, and no log is displayed.

Any idea how I could get this done ?

Thanks.

DLu, 2018-05-14:
This is probably more suited for answers.ros.org, but...

The URDFClient assumes that meshes in your urdf will take the form of `package://name_of_package/mesh_path.stl`. It will [strip the package:// off](https://github.com/RobotWebTools/ros3djs/blob/develop/src/urdf/Urdf.js#L46) and then try to [join it with the path you pass in](https://github.com/RobotWebTools/ros3djs/blob/develop/src/models/MeshResource.js#L33) and load that URL. So the correct answer depends on where your meshes are actually located and whether the MeshLoader will be able to take `file://` based path. 

For instance, if your meshes are stored in `/home/myprofile/catkin_src/src/name_of_repo/name_of_package/mesh_path.stl` the path you would need to pass in would include `name_of_repo`.

If you have additional questions, I'd follow up on answers.ros.org.

closed 2018-05-17 by rctoris

renarded, 2018-05-18:
Thanks for the response.

In fact I got it working on electron.

The test I made was directly on a web browser, and you just can't get access to the file system from firefox or chrome (security safety). (for those interested to read more, [here](https://answers.ros.org/question/216311/ros3djs-not-able-to-display-my-3d-files/)'s a question on answers.ros.org that is similar to this question)