RobotWebTools/ros3djs #227: how to realize 2d pose estimate and 2d nav goal?

Ydcool, 2018-07-03:
or how to drag an arrow out in ros3djs like rviz does?

Ydcool, 2018-10-31:
> any luck with it?

not yet ; ( 
I had change the way to interact with ros3d map for mobile page.

rvipin17, 2018-11-01:
do you have any idea how to do pose estimate in ros3djs with a mouse click, just like we do in ros2djs?

Ydcool, 2018-11-06:
> do you have any idea how to do pose estimate in ros3djs with a mouse click, just like we do in ros2djs?

not much, maybe you should learn more about three.js

amy-fang, 2020-05-15:
Is there anything new here ?

Robotec44, 2021-03-13:
Hi Guys 
Any updates in this case?
I know I can add this functionality with three.js  Raycaster object but I need some help with it. 
Some example or maybe code place where I can start
@rctoris  @DLu @T045T @jihoonl @J-Rojas

MerAARIZOU, 2021-03-14:
@Robotec44 I suggest that you take a look at the Interactive Markers. I used them to manipulate a 7 DoF robot arm from a web page. From their 3D poses it was possible to calculate the IK and send the new path to the robot. [Here](http://wiki.ros.org/ros3djs/Tutorials/VisualizingInteractiveMarkers) is the tutorial about Interactive Markers in ROS3DJS

Robotec44, 2021-03-16:
Thanks I check

closed 2023-02-10 by MatthijsBurgh