RobotWebTools/ros3djs #241: Angular 7 and Viewer, undefined is not a constructor

AndrewJSchoen, 2018-12-03:
We are using Angular 7, and are encountering an issue where the Viewer is not being imported correctly. In our component where we are loading it, we have the following:

```
import { Component, OnInit } from '@angular/core';
import * as ROS3D from 'ros3d';
import * as ROSLIB from 'roslib';

@Component({
  selector: 'environment',
  templateUrl: './environment.component.html',
  styleUrls: ['./environment.component.css']
})
export class EnvironmentComponent implements OnInit {

  private environment: ROS3D.Viewer;

  constructor() { }

  ngOnInit() {
      this.environment = new ROS3D.Viewer({
          divID: 'environment',
          width: 400,
          height: 300,
          antialias: true
      });
  }

}
```
We get the following error: 

```
TypeError: undefined is not a constructor (evaluating 'new ros3d__WEBPACK_IMPORTED_MODULE_1__["Viewer"]({ divID: 'environment', width: 400, height: 300, antialias: true })')
```

Seems like the viewer is not being imported correctly. Is there a fix for this?

LdwgWffnschmdt, 2019-01-09:
Same problem occurs in a Vue.js app. Output says:
`TypeError: ros3d__WEBPACK_IMPORTED_MODULE_0__.viewer is not a constructor`

closed 2023-02-10 by MatthijsBurgh

MattiaDeiRossi, 2023-08-06:
I have the same issue, 

Any update?