RobotWebTools/ros3djs #249: PointCloud2 is not displayed correctly

cmperezg, 2019-01-24:
Hello
I've built an app to display sensor data from a robot, but I can't get PointClouds to display correctly. LaserScans work without issue, so I'm quite sure the structure of my code is fine. This is the configuration I am trying to display:

```
{
        "type" : "Grid",  
        "config" : {
            "size" : 2,
            "color" : "#707070",
            "lineWidth" : 0.01
        }
    },
    {
        "type" : "PointCloud2",  
        "config" : {
            "topic" : "/camera_left/depth/points",
            "material" : { "size": 0.3, "color": "#e8a010" }
        }
    },
    {
        "type" : "PointCloud2",  
        "config" : {
            "topic" : "/camera_right/depth/points",
            "material" : {"size": 0.3, "color": "#e87c10" }
        }
    }
```
The tfClient is added later, and the config is handled by a parser.

However, the point cloud is displayed very weirdly. here is a screenshot:  

[image]

Here is an rviz screenshot of the same rosbag playing, subscribing to the exact same topics:
[image]

Why is it being flattened out this way? Am I missing something in the configuration? Is it even possible to display more than one point cloud at once? I had no issues displaying several laser scans.
After a few seconds this starts to lag like crazy too, but that is a separate issue and I see there is a ticket for it already.

Any help with this issue would be greatly appreciated.

mvollrath, 2019-01-24:
Can you paste the contents of the "fields" (PointField list) for a frame of this cloud?  We've seen issues with some non-standard field layouts.

cmperezg, 2019-01-25:
Hi @mvollrath , thanks for your reply. 

Here are the fields for a frame:
```
- 
  name: "x"
  offset: 0
  datatype: 7
  count: 1
- 
  name: "y"
  offset: 4
  datatype: 7
  count: 1
- 
  name: "z"
  offset: 8
  datatype: 7
  count: 1
---
```

cmperezg, 2019-01-31:
This was not a bug. The problem was that I was not aware that a point limit was being set by default. setting max_pts to 200000 fixed the problem for me.

Kad91, 2019-07-30:
Another question About PointCloud2. It looks like that when `max_pts` (name it N) is selected, only the N first points are displayed. Is there a way to display N points choosing them ( for example every 2 points ) ?

mvollrath, 2019-07-31:
@Kad91 Use [`pointRatio`](https://github.com/RobotWebTools/ros3djs/blob/2f858cfe1c481e61968cb2fb1812ec15e5a1e6f2/src/sensors/PointCloud2.js#L54) for that kind of subsampling.