RobotWebTools/ros3djs #263: Demos do not work in Mobile Safari

anton-matosov, 2019-06-02:
Neither one of demos work in Mobile Safari (iOS 12.3.1)
I see black screen with coordinates frame. I can rotate that frame using gestures, but there is nothing else visualized in any demo.

MerAARIZOU, 2019-06-03:
Have you tried it from another browser? what are the errors from your browser console? What topic are active? We need more information about your issue!

imar759, 2019-06-17:
None of the live [demos](http://robotwebtools.org/demos.html) work properly for me either. I've tried on desktop chrome, firefox etc. on different systems and the same problems occur.
[image]
At first I thought it was to do with my browser extensions but even on a fresh browser they don't work properly.

MerAARIZOU, 2019-06-17:
it seems to me that there is a connection issue with robotwebtools.org It's either the link is dead or there is no websocket (rosbridge) in that server. The solution is that you create your own server side on your localhost (tf2_web_republisher, rosbridge, joint_state_publisher, robot_state_publisher) launch any robot model with gazebo, and publish the js file (ros3d, roslib, collada, etc.) and the 3D model on a apache server, or anyother HTTP server, and you will have your model on your web browser.

jihoonl, 2019-06-17:
Sorry guys. The demo instances are running at the moment due to the cloud instance maintenance issue. I suggest running the examples on your local machine to see how the library works. If someone creates a docker instance that runs demo, I can happily revive the demos on the website.