RobotWebTools/ros3djs #264: Trying to understand why SceneNode objects are created for every message

J-Rojas, 2019-07-10:
Hi, I'm attempting to visualize markers with smooth motion animations. I noticed that for every ROS message, a new SceneNode object is created for all Object3d objects. Visibility is also reset to false for every message received. This causes the objects to become invisible until TF nodes are published as well. This leads to flickering in motions animations that is not smooth.

Why doesn't the implementation reuse the SceneNodes and just update the original objects in the scene directly? Any rationale for this?

mvollrath, 2019-07-10:
I'm not familiar with the ros3d logic for markers, but do recall that markers in ros3d were not working the same as rviz.  I see a `MarkerArrayClient` change in #262 which might be relevant.

J-Rojas, 2019-07-18:
I don't think those checkins address the issue. The markers shouldn't become invisible between TF updates. For example the web republisher also throttles tf updates and will only send updates to the browser when a tf changes vs tf typically being published constantly on a local system. That's probably the difference between ros3d and rviz. I would classify this as a bug, if you agree.

lisaeb, 2020-12-08:
Anything new here? I'm having the same issue when creating one object (tried with `MarkerClient` and `MarkerArrayClient`). However as soon as I have more than one object my objects stay visible.
I also traced this back to the visibility initially set to false in `SceneNode`. If there is more than one object the visibility is set to `true` afterwards by `SceneNodes`' TF handler `tfUpdate`. This function is passed to the[ `subscribe` function of the `TFClient` object](http://robotwebtools.org/jsdoc/roslibjs/current/tf_TFClient.js.html#line163) in roslibjs. As it can be seen there it is only called if `this.frameInfos[frameID].transform` exists. Apparently the transform is calculated after subscribing to the TFClient so the very first message received would not be shown but the second one would. 
Before the new `SceneNode` e.g. for a [MarkerClient](http://robotwebtools.org/jsdoc/ros3djs/current/markers_MarkerClient.js.html#line73) is created the old node [unsubscribes the TFClient](http://robotwebtools.org/jsdoc/roslibjs/current/tf_TFClient.js.html#line192). If there was only one object in ´this.frameInfos[map]´ which is removed due to unsubscribing the whole this.frameInfos[frameID] object is deleted as it empty then. Therefore also the calculated transform, which means the `tfUpdate` function from above is never called if only one object should be shown. With more than one object this function is never emptied which means calculated transforms are not deleted, `tfUpdate` called and the visibility is set to true.
I found two ways to work around this problem: Either setting the visibility in the `SceneNode` to true or preventing empty `this.frameInfos[frameID] objects from being deleted.