RobotWebTools/ros3djs #268: I need to get the camera object and sence object !

BGMjie, 2019-07-18:
My boss ask me to implement a new function of using a slider to control zoom in or zoom out .
[image]
The final pic just like above .
I have seen there is a function (ROS3D.OrbitControls)  can do that ,
But it require 2 parameter (camera and sence ) , it define in ROS3D.Viewer but I don't know how to use it in myself JavaScript file , Please help me ! Thanks a lot !

amy-fang, 2020-05-15:
Is there anything news here?

BGMjie, 2020-05-15:
> Is there anything news here?

 I have change the is library put the interface out

amy-fang, 2020-05-15:
> 
> 
> > Is there anything news here?
> 
> I have change the is library put the interface out
Which library can zoom out it when initialization? My map is bigger and the viewer is small.

BGMjie, 2020-05-15:
[image]

Robotec44, 2021-03-18:
Is there anything news here?