RobotWebTools/ros3djs #269: Use our own URDF

Kad91, 2019-07-23:
Is there a way to use our own URDF with ros3djs ? If yes, maybe we could add another tutorial with all the details about using our own URDF .

MerAARIZOU, 2019-07-23:
You can use whatever URDF you want with ros3djs. Of course if it is well formed and has no errors. If you can use it on gazebo, or visualize it on RViz, you can certainly use it with ros3djs. The usual tutorial is sufficient. What is precisely your issue?

Kad91, 2019-07-23:
URDF works on RViz, for ros3djs, what must be written for the `path` in the `ROS3D.UrdfClient`. And like the tutorial for pr2, do we have to run some lines to make everything work ?

MerAARIZOU, 2019-07-23:
You have to put first your robot_description folder (the one which contains your robot meshes) in a HTTP server (Apache for example), then put the url of your folder in the `path`. For example if you put robot_description in the root of Apache server the `path` would be:
`path : 'http://10.3.66.109/'` 10.3.66.109 being the address of your Apache server. 
The only lines you should pay attention to are:
`frameID` in `ROS3D.UrdfClient` which is the root frame of your urdf model
`fixedFrame` in `ROSLIB.TFClient` which is the root frame of all you TF tree