RobotWebTools/ros3djs #271: Support panning along the scene's XY plane

J-Rojas, 2019-07-24:
The panning behaviour is currently implemented to move the user along the camera's view plane. It would be more useful to pan along the scene's XY plane. This is how RVIZ implements panning and Three.js also has support using 'screenSpacePanning = false'.

arcoyk, 2020-01-29:
@J-Rojas Not sure 'OrbitControls.screenSpacePanning = false' is a perfect solution. I assume Google Earth-like controls are expected as seen in [here in stackoverflow](https://stackoverflow.com/questions/26185074/three-js-orbitcontrols-js-pan-camera-parallel-to-ground-plane-like-google-e). I'll take a look at it and possibly request a pull later.

J-Rojas, 2020-01-29:
It's works great for me.