RobotWebTools/ros3djs #277: marker cannot be deleted

oka1125, 2019-08-12:
I can't delete the marker by publishing Marker message (action=Marker.DELETE) in ROS(Ubuntu).
On the other hand, it is confirmed that the marker was deleted in ROS (Ubuntu).

### note
* I use ROS3D.MarkerClient in ros3djs.
    * The marker is deleted successufully in rviz in ROS(Ubuntu).
* When I use MarkerArray, I can both add and delete it.

oka1125, 2019-08-12:
I found a bug in MarkerClient.process function in ros3d.js.
This function always add a new marker regardless of action (Marker.ADD, Marker.DELETE, etc.).

On the other hand, MarkerArrayClient class does not have such a bug.

peterrw, 2021-03-29:
It also doesn't support "DELETEALL" yet. I will create a pull-request after testing my code.

MatthijsBurgh, 2023-02-10:
Has been fixed in the meantime

closed 2023-02-10 by MatthijsBurgh