RobotWebTools/ros3djs #286: Is there a way to keep the axes displayed

WhatWhatz, 2019-09-06:
Is there any way to keep the axis or frame always displayed. I looked really quick to the code and didn't see any option that controls my problem in OrbitControls.js where the axes are displayed for 1 second only.

WhatWhatz, 2019-09-09:
I've modified the src files and got it to work as an option that you can add when you create the ROS3D.Viewer object. Should I post my modified code here in case someone needs it ?

MerAARIZOU, 2019-09-09:
I personnaly would like to use it. I would appreciate if you share your
version of code. Thank you very much

WhatWhatz, 2019-09-10:
In the zip file you will notice that I added an axesDisplay option that you can configure when initializing the `ROS3D.Viewer`. So adding this line `axesDisplay : true` when creating `ROS3D.Viewer` will keep the axes always displayed and will ignore the `displayPanAndZoomFrame` option. I have rebuilt the library under the `build` directory in case you want to use only the compressed `ros3d.min.js` file.
[ros3djs_withAxesDisplayOption.zip](https://github.com/RobotWebTools/ros3djs/files/3594758/ros3djs_withAxesDisplayOption.zip)
If you notice any problems please let me know. 
It would be awesome if someone could guide me and help me make a pull request so that this option is added to the develop build.