RobotWebTools/ros3djs #293: URL errors in tutorial examples

petroslj, 2019-10-31:
Example HTML code that references external URL resources seems to have numerous errors in both java script URLs and PR2 model URDF.  Context:
     Ubuntu 18.04 / ROS Melodic
     Tutorial 1: http://wiki.ros.org/ros3djs/Tutorials/VisualizingAURDF
     Tutorial 2: https://github.com/PacktPublishing/ROS-Robotics-Projects.git
Examples in both tutorials have bad java script URLs and cannot find & load PR2 URDF files.
Typical errors:
GET https://static.robotwebtools.org/threejs/current/STLLoader.min.js net::ERR_ABORTED 403
GET http://resources.robotwebtools.org/pr2_description/meshes/base_v0/base.dae net::ERR_NAME_NOT_RESOLVED

A screen shot of the javascript console errors is here:
[image]

msfarhan, 2020-01-17:
im also facing same issue . i think path in urdf client which refers to resources.robotwebtools.org doesnt exist and hence we cannot load mesh .. im also looking for answers

chrisfernam2, 2020-09-01:
Hi any luck resolving this?

closed 2023-02-07 by MatthijsBurgh