RobotWebTools/ros3djs #310: Problem with all examples in ros3djs repo on localhost

martin-pc, 2020-03-23:
I am currently running Ubuntu 18.04 and ROS melodic.
Currently, I am not able to get any of the examples using chrome or fixefox.
Below are the steps I follow

1. I checkout the ros3Djs repo
2. follow the guide for setup and building.
3. Then I start the rosbridge_websocket
4. In chrome or firefox i load interactivemarkers.html
Below are the error I get from the rosbridge_websocket:
```
process[rosbridge_websocket-2]: started with pid [27398]
process[rosapi-3]: started with pid [27399]
2020-03-23 19:36:31+0100 [-] Log opened.
2020-03-23 19:36:31+0100 [-] registered capabilities (classes):
2020-03-23 19:36:31+0100 [-]  - rosbridge_library.capabilities.call_service.CallService
2020-03-23 19:36:31+0100 [-]  - rosbridge_library.capabilities.advertise.Advertise
2020-03-23 19:36:31+0100 [-]  - rosbridge_library.capabilities.publish.Publish
2020-03-23 19:36:31+0100 [-]  - rosbridge_library.capabilities.subscribe.Subscribe
2020-03-23 19:36:31+0100 [-]  - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
2020-03-23 19:36:31+0100 [-]  - rosbridge_library.capabilities.advertise_service.AdvertiseService
2020-03-23 19:36:31+0100 [-]  - rosbridge_library.capabilities.service_response.ServiceResponse
2020-03-23 19:36:31+0100 [-]  - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
2020-03-23 19:36:31+0100 [-] WebSocketServerFactory starting on 9090
2020-03-23 19:36:31+0100 [-] Starting factory <autobahn.twisted.websocket.WebSocketServerFactory object at 0x7f444fa6e3d0>
2020-03-23 19:36:31+0100 [-] [INFO] [1584988591.489213]: Rosbridge WebSocket server started at ws://0.0.0.0:9090
2020-03-23 19:36:41+0100 [-] failing WebSocket opening handshake ('WebSocket connection denied: origin 'file://' not allowed')
2020-03-23 19:36:41+0100 [-] dropping connection to peer tcp4:127.0.0.1:43346 with abort=False: WebSocket connection denied: origin 'file://' not allowed
2020-03-23 19:44:16+0100 [-] failing WebSocket opening handshake ('WebSocket connection denied: origin 'null' not allowed')
2020-03-23 19:44:16+0100 [-] dropping connection to peer tcp4:127.0.0.1:43396 with abort=False: WebSocket connection denied: origin 'null' not allowed
```
Chrome creates the 'file://' and Firefox the 'null' response.
I am currently missing something here, but I do not know what.

Janphr, 2020-03-30:
Exact same problem with ros2djs.

prefpkg21, 2020-04-19:
You need to either enable local file loading in Chrome or  run a web server to host/serve the files on the robot

mvollrath, 2020-04-20:
This is potentially due to a bug in rosbridge preventing local file origins from working.  Should be fixed by https://github.com/RobotWebTools/rosbridge_suite/pull/469 which will be in the next rosbridge release.