RobotWebTools/ros3djs #319: urdf.html example not working.

Jhala420, 2020-07-09:
I'm running ros3djs urdf.html example. But every time I got the following error. I search on google but I don't find any solution. 

[image]

The output of other commands are the following.

[image]
[image]

[image]
[image]
[image]
I have experience with ROS but I'm new to Robot Web Tool. 
ros2djs, roslibjs are working perfect.

I'm Using ROS Melodic.
Python 2.7.
Chrome Version 83.0.4103.116 (Official Build) (64-bit)

MerAARIZOU, 2020-07-09:
Are you using a HTTP server? I see that you are opening HTML files in your browser. Try to install Apache server and put your HTML file and all the JS files then fix the URLs on your code

Jhala420, 2020-07-10:
Thanks.
Problem Solved.

closed 2020-07-10 by Jhala420