RobotWebTools/ros3djs #322: MarkerArrayClient out of memory

chenfubiao, 2020-07-20:
Call 'MarkerArrayClient' and the memory will keep rising until it runs out of memory.

chenfubiao, 2020-07-21:
I found the reason，The texture not dispose.

roddc, 2020-07-27:
@chenfubiao Hi, could you show me how to use MarkerArrayClient? I am trying to use this API to show "trajectory_node_list", but nothing displayed on the viewer

chenfubiao, 2020-07-30:
> @chenfubiao Hi, could you show me how to use MarkerArrayClient? I am trying to use this client to show "trajectory_node_list", but nothing displayed on the viewer

    var ros = new ROSLIB.Ros({
       url : 'ws://localhost:9090'
    });

    var viewer = new ROS3D.Viewer({
      divID : 'viewer',
      width : window.innerWidth,
      height : window.innerHeight,
      antialias : true,
      cameraPose : { 
        x: 0,
        y: 0,
        z: 80
      }
    });

    // Setup a client to listen to TFs.
    var tfClient = new ROSLIB.TFClient({
      ros : ros,
      angularThres : 0.01,
      transThres : 0.01,
      rate : 10.0,
      fixedFrame : '/camera_link'
    });

    var markers = new ROS3D.MarkerArrayClient({
      ros: ros,
      tfClient: tfClient,
      rootObject: viewer.scene,
      topic: '/your_topic'
    });

 if you have to install  rosbridge_server and tf2_web_republisher

Techming, 2022-06-15:
Hi @chenfubiao how did you work around with the texture?