RobotWebTools/ros3djs #331: 3D object and map cost lots of CPU and Memory

amy-fang, 2020-08-28:
I draw a map and 3D object like in rviz. When I run it and play a bag, the browser cost lots of CPU and Memory. I played the bag two times. It always cost CPU 100%, and memory growing. See below.

First time played the bag.
First pic is just started playing. Web content cost **CPU 102.4% Memory 14%**.
[image]

Second pic is about half played. Web content cost **CPU 145.1% Memory 24.3%**.
[image]

Third pic is finished the first time. Web content cost **CPU 90.8% Memory 41.4%**.
[image]

After about 10min I played second time.
First pic is just started playing. Web content cost **CPU 98.2% Memory 46.3%**.
[image]

Second pic is about half played. Web content cost **CPU 100.5% Memory 49.4%.**
[image]

Third pic is finished the second time. Web content cost **CPU 104.4% Memory 60.7%**.
[image]

The memory is growing until I close the browser.Is the resource not released?

I used multiple 3D object, list following.
ROS3D.Grid
ROS3D.Viewer
ROS3D.OccupancyGridClient
ROS3D.Odometry
ROS3D.LaserScan*2
ROS3D.PointCloud2
ROS3D.Path*7
ROS3D.Polygon*3
ROS3D.Urdf

It's terrible. Can someone help me?

Rezenders, 2021-05-27:
Found a solution for this? I am experiencing the same problem, memory keeps growing indefinitely.

amy-fang, 2021-05-28:
> 
> 
> Found a solution for this? I am experiencing the same problem, memory keeps growing indefinitely.

I removed all point cloud and laser scan. Only keep the important path.Also removed urdf, use polygon to represent the vehicle. That can keep the memory within a reasonable range .