RobotWebTools/ros3djs #338: OrbitControls.userRotate=false doesn't work

sdtech-kawabatahitoshi, 2020-09-15:
I tried the following code to disable the rotate operation by the users.
(```viewer.cameraControls``` is the instance of ```OrbitControls```)

```
    var ros = new ROSLIB.Ros({
      url : 'ws://192.168.10.133:9090'
    });

    var viewer = new ROS3D.Viewer({
      divID : 'map',
      width : 800,
      height : 600,
      antialias : true,
      cameraPose : {x : 0, y : 0, z : 10}
    });

    viewer.cameraControls.userRotate = false;  // <-- This should disable userRotate
```

However, the result is:
* When I mousedown and mousemove, rotation starts
* When I mouseup and mousemove, rotation continues and never stops

The functions of ```onMouseDown()``` and ```onMouseUp()``` in ```OrbitControls.js``` are as follows. If ```userRotate``` is set false, ```state``` should not be changed to ```STATE.ROTATE```, and rotation should not be started.

```
  function onMouseDown(event3D) {
    var event = event3D.domEvent;
    event.preventDefault();
    switch (event.button) {
      case 0:
        state = STATE.ROTATE;    // <-- Before this, "if (!that.userRotate) break;" should be added 
        rotateStart.set(event.clientX, event.clientY);
        break;
    ...
```

```
  function onMouseUp(event3D) {
    if (!that.userRotate) {    // <-- This if statement is not needed
      return;
    }
    state = STATE.NONE;
  }
```

RMStonight, 2023-01-10:
I tried to set `viewer.cameraControls.userRotateSpeed = 0`.

MatthijsBurgh, 2023-01-10:
> I tried to set `viewer.cameraControls.userRotateSpeed = 0`.

And did it work?

RMStonight, 2023-01-14:
> 

yes

RMStonight, 2023-01-14:
> > I tried to set `viewer.cameraControls.userRotateSpeed = 0`.
> 
> And did it work?

Yes, now i can't rotate the camera.

closed 2023-02-07 by MatthijsBurgh