RobotWebTools/ros3djs #342: Robot model update is getting slower over time

tandukion, 2020-10-12:
I tried to run a continuous looping motion on my system for a long-run test.
The robot is a Yaskawa Motoman robot, and the motion is a pick and place motion.
The system launches both tf2_web_republisher for ros3djs to show the visualization on a web page, and also RViz as the default visualization.

After some period, the robot web visualization (ros3djs) and RViz became different, as the web visualization getting slower in visualizing the update on the robot model.

I don't know whether this is an issue for ros3djs or roslibjs, but since the issue affects the visualization I suppose it should be for ros3djs.

J-Rojas, 2020-10-17:
How long were you running your visualization? How many tf updates were being sent? What versions of these modules are you using?

tandukion, 2020-10-20:
@J-Rojas I ran it for 1 day, but I think the problem was already started occuring after 12 hours.

The tf updates were for 6 links. I set the tf client rate to 30.0 (I need to make it as smooth as RViz if possible).

I could not verify the version, since I used the library based on the [tutorial](http://wiki.ros.org/ros3djs/Tutorials/VisualizingAURDF) on ROS wiki page. The ros3d.js that I used is from the static page [link](https://static.robotwebtools.org/ros3djs/current/ros3d.js). Is it so much older than the latest build on this repo?

closed 2023-02-07 by MatthijsBurgh