RobotWebTools/ros3djs #347: OccupancyMap and OccupancyMapClient are not being built

AliKarpuzoglu, 2020-11-18:
Hi,
even after adding 

```
export * from './navigation/OccupancyMap'
export * from './navigation/OccupancyMapClient'
```

to `es6-support/index.js`

the grunt build fails with an "unexpected token" 
```
Warning: Command failed: rollup -c

src-esm/index.js → ./build/ros3d.js...
[!] (commonjs plugin) SyntaxError: Unexpected token (59:35) in /PATH/ros3djs/src-esm/navigation/OccupancyMap.js
src-esm/navigation/OccupancyMap.js (59:35)
```

is the file not finished? the weird thing is that the issue only occurs after the es6 output

psaripp, 2020-12-16:
It seems the issue is it fails to generate classes from the es5 code. It's responsible by `es6-transpiler`, but I could not understand how the code supposed to be formatted or written to be satisfied by that regex madness there. If someone with a better understanding of that may explain it to me, I'm happy to reformat these parts to satisfy this thig.

AliKarpuzoglu, 2021-01-06:
@psaripp I am wondering: how did you get the code to run if they didn't get included in the js? Are you using a different building method?

brean, 2021-03-11:
After converting the code to ES5 and giving every class its own js-file I was able to transpile it, but it is hardcoded to work with the message type "octomap_msgs/Octomap" only, not (in my case) "nav_msgs/OccupancyGrid", so it does not work as OccupancyGridMap which I would have expected...

I can create a pull request providing my fixes but I think this needs some more discussion and development, I just made stuff work without knowing what it should do.

----

I also suggest to switch to ES6 for the whole codebase of ros3djs instead of using this magic to convert the old ES5 to ES6-style but I know that this potentially means a lot of work.
Because we are using THREE.js/WebGL we need a modern browser and can assume ES6 anyway IMHO.