RobotWebTools/ros3djs #356: LaserScan does not display the points, ros3d.js:21217

amoula20, 2021-01-17:
Hello,

I'm having a problem displaying the laser scan on my browser. here is my code:
   `   var viewer = new ROS3D.Viewer({
      divID : 'scan',
      width : 800,
      height : 600,
      background: '#444444',
      antialias : true
    });

    var tfClient = new ROSLIB.TFClient({
      ros : ros,
      angularThres : 0.01,
      transThres : 0.01,
      rate : 10.0,
      fixedFrame : ''//laser_base_link
    });

    var scanclient = new ROS3D.LaserScan({
        ros: ros,
        topic: '/scanR',
        tfClient: tfClient,
        rootObject: viewer.scene,
        material: { size: 0.05, color: 0xff00ff }
    });
`
I'm running the tf2_web_republisher, and I have this message on the console:
THREE.WebGLRenderer 89                        ros3d.js:21217

city945, 2022-06-01:
Hi, I had the same problem. Did you solve it？

MatthijsBurgh, 2023-02-07:
Stale

closed 2023-02-07 by MatthijsBurgh