RobotWebTools/ros3djs #400: Best practices for ROS websocket objects?

aamirhatim, 2021-06-03:
Let's say I have a 3D viewer that displays lidar data for my robot. Now I want to create a second 3D viewer that shows a map that is being generated by a package like gmapping. What would be a recommended best practice - a single ROSLIB.Ros() object that gets passed to both viewers' visualization components, or a separate ROSLIB.Ros() object for each viewer? I currently am implementing the first option but since I'm quite new to web dev and RWT libraries I wanted to reach out to the community for any thoughts on this.

MatthijsBurgh, 2021-06-04:
I think it is best practice to reuse the ROS object.

aamirhatim, 2021-07-05:
What is your reasoning behind using the single ROS object?

MatthijsBurgh, 2021-07-06:
Easier to debug; Single instance which maintains a connection, etc.

closed 2023-02-10 by MatthijsBurgh