RobotWebTools/ros3djs #440: Memory leaks into OccupancyGridClient

letis33, 2021-09-18:
The current grid object is not removed from root object into OccupancyGridClient.processMessage() function.
=> only from sceneNode

Memory is growing very fast... 

```
processMessage(message){
  // check for an old map
  if (this.currentGrid) {
	  // check if it there is a tf client
	  if (this.currentGrid.tfClient) {
	    // grid is of type ROS3D.SceneNode
	    this.currentGrid.unsubscribeTf();
	  }

**==>**	  this.sceneNode.remove(this.currentGrid);

	  this.currentGrid.dispose();
  }
```

- Library Version: latest (develop branch)
- ROS Version: Noetic
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**

Launch continuousmap.html

**Expected Behavior**

mrdoob.github.io/stats.js/build/stats.min.js shall have memory stable
=> about 150MB in my case

**Actual Behavior**

Memory is growing until Chrome freeze
=> several giga bytes

**Quick fix**

this.rootObject.remove(this.currentGrid); instead of this.sceneNode.remove(this.currentGrid);

MatthijsBurgh, 2021-09-20:
I think your solution is not complete. Because of https://github.com/RobotWebTools/ros3djs/blob/64ba1a2af2b72063521589b8591871d245ccda64/src/navigation/OccupancyGridClient.js#L86-L103

When `tfclient` resolves to false, I think `this.rootObject.remove(this.currentGrid)` is correct, but when `tfclient` resolves to true, `this.sceneNode.remove(this.currentGrid)` is correct. Do you agree?

letis33, 2021-09-20:
Hi,

I'm totally agree with you.

Thanks for your reply.

commit f39378d3 (2021-11-07): Fix Memory leak of #440 (#444)

* Add missing semi-colon

* Fix #440, correctly delete old grid
closed 2021-11-07 by MatthijsBurgh