RobotWebTools/ros3djs #465: Array buffer allocation failed

Jaetriel, 2021-11-09:
**Description**

I'm trying to create a webapp to display a 3d map being built in real-time with data coming from a LiDAR sensor connected to ROS, running on another machine and transmitting the point cloud data via websocket.  The webapp runs fine for a while, and at some point after refreshing the page a few times or clearing the scene and loading a few different clouds I start getting the 
`Array buffer allocation failed at new ArrayBuffer (<anonymous>) at new Float32Array (<anonymous>) at Points.setup` and the point cloud stops being rendered.

- Library Version: 1.01
- ROS Version: Melodic
- Platform / OS: NVidia Jetson Xavier AGX / Jetpack OS based on Ubuntu 18.04(arm64 cpu architecture)

**Steps To Reproduce**

I haven't been able to find a concrete sequence of steps that get me the error every time, but after running the webapp for a little while (sometimes 15 min sometimes an hour) and rendering point clouds, it inevitably shows up.

Jaetriel, 2021-11-09:
@MatthijsBurgh sorry I'm not sure I know how to check exactly how much memory the webapp specifically is using?

MatthijsBurgh, 2021-11-09:
It depends on your browser. Please use Google

Jaetriel, 2021-11-10:
So, I'm still not sure if I used the correct tool but I used the Chromium task manager, and I checked the memory footprint of the process. While running the slam on my host PC and sending data to the webapp, the memory kept increasing, which looks to be normal. I then paused the algorithm and the webapp stopped receiving data, but the memory slowly kept increasing as long as i had this tab highlighted. I refreshed the page but that didn't clear all the memory used. I even cleared the scene with `scene.remove.apply` but that buffer seems to not be clearing. At this point I think it's worth mentioning I am using the decay option for PointCloud2 from following this PR: #357

MatthijsBurgh, 2021-11-10:
What happens when you don't use the decay option

Jaetriel, 2021-11-23:
@MatthijsBurgh i'm sorry for taking so long to respond, for now I think it might be best to close this issue because I am having trouble reproducing the bug, and I might have to leave it until such time when I can focus more on finding a way to reproduce it

closed 2021-11-23 by MatthijsBurgh