RobotWebTools/ros3djs #477: rviz is normal ros3djs fails

vusdo, 2021-12-23:
**Description**
rviz is normal ros3djs fails

- Library Version: 0.17.0-SNAPSHOT
- ROS Version: melodic
- Platform / OS: ubuntu 18.04

**Steps To Reproduce**

**Expected Behavior**
[image]

**Actual Behavior**
[image]

MatthijsBurgh, 2021-12-23:
As your `steps to reproduce` are empty and you don't provide any information about what you are doing. I can't help you based on two screenshots.

vusdo, 2021-12-27:
> 당신 `steps to reproduce`은 비어 있고 당신이하고있는 일에 대한 정보를 제공하지 않습니다. 두 개의 스크린샷으로는 도움을 드릴 수 없습니다.

I have edited the article for you.
Please review.

VimalGohel1, 2022-01-27:
I'm also facing the same issue. I've observed that when the number of marker points increases, the 3D viewer is not able to handle correct 3D space rending and missing points.

Can we have an update on when can we have a possible resolution for this bug?

MatthijsBurgh, 2022-01-27:
I have not been able to take a look at this.

If you want to speed up the process, please prepare a example repo/package, which reproduces the issue. So that I can install, (build) and run it. That would really reduce the startup effort on my side.

VimalGohel1, 2022-02-09:
Hello @MatthijsBurgh, I've prepared the package as requested.
[Marker_trial_sample.zip](https://github.com/RobotWebTools/ros3djs/files/8032518/Marker_trial_sample.zip)

We have also observed additional issues with the library:
1. Blinking issue - Data is displayed, but vanishes abruptly. (You can see in the sample provided)
2. Memory - The browser hangs up after sometime (During performance check, memory consumption increases every time the data is received on the topic)

Thanks.

MatthijsBurgh, 2022-02-09:
A marker message is only shown up to 1250 points per message. See
https://github.com/RobotWebTools/ros3djs/blob/5d40a756036bfbf2f9742dee5b98c7dcf1fd738c/src/markers/Marker.js#L169
and
https://github.com/RobotWebTools/ros3djs/blob/5d40a756036bfbf2f9742dee5b98c7dcf1fd738c/src/markers/Marker.js#L200

This is the current behaviour.  If you want to change that behaviour, please open a PR. I think that the 1250 should still be the default, but that it is configurable.

For the blinking: As you are publising a new Marker in the same NS and with the same ID. The old one is removed. As it might take some time before it renders a new Marker. There could be some time without any markers.

For the memory issues: There have been some memory fixes since the last release. I run your example with the with roslibjs 1.2.0 and ros3djs on develop branch. But I don't experience a memory leak. Ofcourse it will use significant memory (500-700MV).

(I will try to get a new release out before the end of the month.)

VimalGohel1, 2022-02-15:
> This is the current behaviour. If you want to change that behaviour, please open a PR. I think that the 1250 should still be the default, but that it is configurable.

We tried modifying this value as you suggested. Apparently, this value is `stepSize` used for incrementing in `for`  loop. It is not trying to limit the number of points added to the scene. We observed no difference in the output.

`For the blinking`: Thank you. Understood why the blinking occurs. 

`For the memory issues`: Thank you. Will wait for the new release.

MatthijsBurgh, 2022-02-15:
> We tried modifying this value as you suggested. Apparently, this value is `stepSize` used for incrementing in `for` loop. It is not trying to limit the number of points added to the scene. We observed no difference in the output.

So by increasing the stepsize, we skip points. This is done, so we skip points through the array and not all points in the end. As it is very likely that points close to each other in the array are close to each other in 3D space.

MatthijsBurgh, 2022-02-15:
> `For the blinking`: Thank you. Understood why the blinking occurs.

You can fix this by publishing update messages.

VimalGohel1, 2022-02-22:
> So by increasing the stepsize, we skip points. This is done, so we skip points through the array and not all points in the end. As it is very likely that points close to each other in the array are close to each other in 3D space.

Yes. Now we are able to visualize all the points after reducing `stepSize` to a minimum. Thank you for resolving this.

VimalGohel1, 2022-02-22:
> > `For the blinking`: Thank you. Understood why the blinking occurs.
> 
> You can fix this by publishing update messages.

Thank you. Fixed this too. :+1:

MatthijsBurgh, 2022-02-22:
I am closing this issue. Feel free to open a new issue, when you experience other issues.

closed 2022-02-22 by MatthijsBurgh

VimalGohel1, 2022-03-04:
Hi @MatthijsBurgh. Can you let us know when we can expect a new release?

MatthijsBurgh, 2022-03-04:
I hope sometime in march.

VimalGohel1, 2022-03-24:
Hi @MatthijsBurgh, Is there any final deadline when the release can be expected?

If not, could you please share which branch should consider for the latest memory related fixes?

MatthijsBurgh, 2022-03-24:
No, I can't provide a deadline.

The develop branch should do the trick. If that isn't the case, the develop branch contains other memory issues.

VimalGohel1, 2022-03-24:
Thank you,. We will check it out and let you know further