RobotWebTools/ros3djs #554: Broken links to documentation

aamirhatim, 2022-07-14:
**Description**
Looks like the JSDocs to the current 2D and 3D libraries are no longer available. I get a 404 message when trying the links given on the readme pages on Github as well as the hyperlinks on the ROS wiki page.

- Library Version: 0.15.0 and current (ros3d), 0.6.0 and current (ros2d)
- ROS Version: N/A
- Platform / OS: N/A

**Steps To Reproduce**

OPTION 1:
1. Go to [repo page](https://github.com/RobotWebTools/ros3djs) for ros3djs ([for ros2d, go here](https://github.com/RobotWebTools/ros2djs)) on Github
2. Click the JSDoc link in the readme

OPTION 2:
1. Go to [repo page](https://github.com/RobotWebTools/ros3djs) for ros3djs ([for ros2d, go here](https://github.com/RobotWebTools/ros2djs)) on Github
2. Click the ROS Wiki link in the readme
3. Click the "current" hyperlink under the API Documentation section

**Expected Behavior**
Browser navigates to JSDoc link to documentation

**Actual Behavior**
Error 404 landing page

MatthijsBurgh, 2022-07-16:
Readme and ROS wiki links are incorrect. Check the links on the top right corner of the repo. I will update the README links at some point. Don't have time right now.

aamirhatim, 2022-08-02:
Ah, thanks for that!

MatthijsBurgh, 2022-08-03:
I will reopen this for my own administration

MatthijsBurgh, 2022-08-30:
READMEs are fixed by https://github.com/RobotWebTools/ros3djs/pull/559 and https://github.com/RobotWebTools/ros2djs/pull/107

closed 2023-02-10 by MatthijsBurgh