RobotWebTools/ros3djs #593: There is no <size> defined on THREE, it is changed to <linewidth>

marantos, 2022-12-22:
https://github.com/RobotWebTools/ros3djs/blob/7e61ef5200c0fe0aa8f4612f1a84b1f3c3d759ce/src/markers/Marker.js#L103

MatthijsBurgh, 2022-12-22:
Thanks for this line of code from the library :wink:.

This library uses Three R89 for now. So when the library is incompatible with Three R89, I am willing to help you. I need to skip any compatibility issues with other versions of the Three library.

marantos, 2022-12-22:
It seems that is a bug, at least for my eyes. I mean, that also in ros3d tagged version 1.1.0 that depends on Three r89, there is the same code. So the browser complains about that there is not a size to be assigned.

MatthijsBurgh, 2022-12-22:
@marantos can you provide a link to the three documentation of this class/attribute?

marantos, 2022-12-23:
> @marantos can you provide a link to the three documentation of this class/attribute? Of course to the documentation of release r89.

Check the doc [here](https://github.com/mrdoob/three.js/blob/r89/docs/api/materials/LineBasicMaterial.html). Sorry that you need to read a raw html. 😊

You can see also the implementation [here](https://github.com/mrdoob/three.js/blob/r89/src/materials/LineBasicMaterial.js) . Tbh, I see also that it puts `linewidth=1`, despite the input.

MatthijsBurgh, 2022-12-27:
@marantos could you please check the fixes on the [`updates_fixes`](https://github.com/RobotWebTools/ros3djs/tree/updates_fixes) branch. Reminder, the build files aren't updated yet. So you still need to recompile.

marantos, 2022-12-27:
Nice @MatthijsBurgh !

marantos, 2022-12-27:
@MatthijsBurgh can I add a commit in this branch in order to fix also #594 .

MatthijsBurgh, 2022-12-27:
@marantos sorry, that is difficult to do. You will no to fork this repo and open a PR to this branch.

marantos, 2022-12-27:
Ok @MatthijsBurgh . I will fork the repo and I will create a PR targeted to this branch. Is that OK?

MatthijsBurgh, 2022-12-27:
@marantos sounds good

closed 2023-01-13 by MatthijsBurgh