RobotWebTools/ros3djs #625: my urdf in rviz not same in ros3d viewer

HESHAM47GAMAL, 2023-07-13:
**Description**

- Library Version:
- ROS Version: noetic
- Platform / OS:

**Steps To Reproduce**

I want to visaulize my robot as appear in rviz

view_3dmap() {

     // Create the main viewer.
     var viewer = new window.ROS3D.Viewer({
        divID : 'urdf',
        width : 800,
        height : 600,
        background : '#EEEEEEE',   
        antialias : true
      });

      // Add a grid.
    viewer.addObject(new window.ROS3D.Grid({
      cellSize : 0.5,
      color : '#0099cc'
    }));

    // Setup a client to listen to TFs.
    var tfClient = new window.ROSLIB.TFClient({
      ros : this.state.ros,
      angularThres : 0.01,
      transThres : 0.01,
      rate : 10.0, 
      // fixedFrame : '/base_link'
    });

     // Setup the URDF client.
     var urdfClient = new window.ROS3D.UrdfClient({
      ros : this.state.ros,
      tfClient : tfClient,
      rootObject : viewer.scene,
      loader : window.ROS3D.COLLADA_LOADER
    });

  }

**Expected Behavior**

[image]

**Actual Behavior**

[image]

MatthijsBurgh, 2023-07-13:
I see react issues. I will not help with that. My time is limited.

So please convert to an example without React. When you provide such an example, I am willing to help you.

HESHAM47GAMAL, 2023-07-14:
thanks for trying you to help me 
but I solve my problem 
my problem that my robot is file.urdf.xacro and start my robot with   <link name="base_link">  that conect last of links and joint to it that is make big box in react ros3d and to make sure that this link don't any problem 
should be written by one way of two 
1. only contain inertial 

    <link name="base_link">

        <inertial>
            <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
            <mass value="1.0"/>
            <inertia ixx="0.0" ixy="0.0" ixz="0.0" iyy="0.0" iyz="0.0" izz="0.0"/>
        </inertial>

    </link>
2. make sure that if use <visual name=""> that box size not equal zero make it very small number as you will not face any problem in rviz but in ros3d react make problem like I face

<link name="base_link">
        
        <visual name="">
            <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
            <geometry>
                <box size="0.01 0.01 0.01"/>
            </geometry>
            <material name="">
                <color rgba="0.0 1.0 1.0 1.0"/>
                <texture filename=""/>
            </material>
        </visual>
        
    </link>

closed 2023-07-15 by MatthijsBurgh